dgomes/homeGW #1: Support esp8266

maragelis, 2018-02-23:
Is there any chance this can work with an esp8266

dgomes, 2018-02-23:
aslong as you provide 5v to the radio receiver, see no reason why it shouldn't work

maragelis, 2018-02-24:
I am giving it 5v, but cannot seam to get it to work. I am using gpio5 (D1) but no data is coming through. Even tried you demos, but still nothing.  On arduino uno everything works fine.

maragelis, 2018-02-24:
Ok got it working had to change the below function 

bool HomeGW::setup(uint8_t pin) {
  

  HomeGW::pin = pin;

  pinMode(pin, INPUT);
  digitalWrite(pin, LOW);

  attachInterrupt(pin, HomeGW::handleInterrupt, CHANGE); // 1 = PIN3

  return true;
}

Esp has interrupts on all pins and names them with pin numbers. Thanks

dgomes, 2018-02-24:
Good to know :blush:

dgomes, 2018-02-24:
So only the test for pin 3 or 2 was failing ?

If you make a PR with a test that covers Arduinos and ESP8266 I will accept it.

maragelis, 2018-02-24:
You where also doing a pin-2 on the attachInterrupt() so that had to change to.

closed 2018-02-24 by maragelis

AndyRL, 2018-03-14:
Hi maragelis.
Can you help me? Like you I have got this to work fine on a Uno - but not an ESP8266.....
Looking at homeGW.cpp there is #if defined(ESP8266)
As far as my knowledge of Arduino is concerned, I should have a #define (ESP8266) in my main prog.
But that throws up an error - macro names must be identifiers
If I include #define ESP8266 nothing seems to change - and I just get garbage.
I am using hass_serial.ino (with the JSON stuff removed)

It would seem you have got this to work (with the fix you posted) so any pointers would be a great help.

BTW - my 433Mhz rx works fine on 3.3v!

Many thanks

maragelis, 2018-03-14:
defined(ESP8266) is an AVR directive built into ESP8266/Arduino 
see https://github.com/esp8266/Arduino/issues/2391

Check out my project for ideas https://github.com/maragelis/433MhzToMqttBridge/blob/master/MhzToMqtt.ino

AndyRL, 2018-03-14:
Hi maragelis.
Thanks for a swift reply.
I;m no arduino expert..... so are you saying if I select ESP8266 in the IDE prior to compile that ESP8266 will be automatically defined?

I will certainly check out your project....
Again - many thanks...

jgracio, 2018-09-17:
Works perfectly on atmelavr but I'm not being able to get reading from the same radio on an ESP8266 platform (wemos d1).

As my arduino/esp8266 knowledge is not deep enough to understand what was writtner here, my question is if i have to replace function setup on homegw.cpp file?

bool HomeGW::setup(uint8_t pin) {
HomeGW::pin = pin;
pinMode(pin, INPUT);
digitalWrite(pin, LOW);
attachInterrupt(pin, HomeGW::handleInterrupt, CHANGE); // 1 = PIN3
return true;
}

Thanks

lvPokka, 2019-09-24:
before attachInterrupt add delay(2000); 
it will fix issue with restart