dgomes/homeGW #10: superheterodyne 433mhz

vaidyasr, 2019-01-12:
Hi,
Could you please suggest a superheterodyne 433mhz which shown like in your schematic? I can found only like this with 4 pins, whereas your diagram shows 8 pins.
https://www.aliexpress.com/item/QIACHIP-3pcs-superheterodyne-433mhz-rf-Wireless-receiver-Module-small-size-low-power-For-433-Mhz-remote/32848494267.html

dgomes, 2019-01-12:
I got mine from farnell.com

It is no longer available, but searching on AliExpress I could easily find a similar one:

https://www.aliexpress.com/item/RXB6-433Mhz-Superheterodyne-Wireless-Receiver-Module-ARM-AVR/32714431287.html?spm=2114.search0104.3.1.6fa12a7e5XhwTg&ws_ab_test=searchweb0_0,searchweb201602_4_10065_10068_319_317_10696_10084_453_454_10083_10618_431_10304_10307_10820_10821_10301_537_536_10059_10884_10887_100031_321_322_10103,searchweb201603_6,ppcSwitch_0&algo_expid=04b82494-3d97-481b-b02a-e7cc3f97a0bd-0&algo_pvid=04b82494-3d97-481b-b02a-e7cc3f97a0bd&transAbTest=ae803_3

vaidyasr, 2019-01-12:
Thank you. And the antenna length is mandatory like you have mentioned? 17.4cm

dgomes, 2019-01-12:
antenna length is mandatory!

vaidyasr, 2019-01-12:
all I was able to find is 15cm like the one here https://www.aliexpress.com/item/NEW-Black-GSM-GPRS-Small-Suction-Cup-Antenna-433Mhz-3dbi-Magnetic-Base-1-5M-RG174-Cable/32786258061.html