dgomes/homeGW #11: ESP8266 with MQTT

vaidyasr, 2019-02-01:
Hi,
I would like to use it with my ESP8266 and HA. The ESP8266 is using Wifi login and password. Still do I need to connect to my Raspberry-pi via USB, to use it in HA with serial connection?. Can I use MQTT to get the data?.

For ESP8266, what will be the connection diagram?. is that same like connecting to Arduino?.

dgomes, 2019-02-01:
Hi,

This is just a library! You need to write your own code to connect via MQTT in the ESP8266.

In the examples dir you find code that exemplifies the use of the library, it connects through a serial interface (that one works with an arduino). If you want to use the ESP8266 and MQTT you need to start fresh.

vaidyasr, 2019-03-23:
Finally I got my components setup. I am trying https://github.com/maragelis/433MhzToMqttBridge/blob/master/MhzToMqtt/MhzToMqtt.ino where using your HomeGW with MQTT. Am using NodeMCU as a gateway. Any idea on what pin I have to connect?. D3 or D2?. Sorry I am a newbie in flashing/setting things up.

dgomes, 2019-03-23:
I don't own/use NodeMCU, so I can't help you there. Have you asked @maragelis ?

maragelis, 2019-03-23:
I use D5 gpio14

vaidyasr, 2019-03-24:
> I use D5 gpio14

Thank you @maragelis 
Do I need to modify your code to use D5?.
No signals received on serial monitor. I was able to receive using Uno per project, but I want to use MQTT.

vaidyasr, 2019-04-04:
Thank you all. working fine.

closed 2019-04-04 by vaidyasr