dgomes/homeGW #12: RC-switch library

danricho, 2019-04-11:
is there any way to allow the [RC-SWITCH](https://github.com/sui77/rc-switch) library to work alongside HomeGW using the same RF receiver?

I have some switches which work well with RC-switch but are unreliable with the generic raw plugin.

dgomes, 2019-04-11:
Unfortunately no, both rely in the interrupt functionality which cannot be shared amongst libraries.

It would take a full rewrite of both libraries to make that possible.

closed 2019-04-11 by dgomes

danricho, 2019-04-11:
I got it working by wrapping both libraries’ interrupt ISR, with a new one which subsequently calls both.

I also tweaked both libraries to make the built in ISRs public.

After running the setup method I detached the interrupt and remapped it to my new ISR.

I think a nice solution might be to accept an argument to the setup function which stops the interrupt attachment.

RC-Switch doesn’t attach until another function is called (I can’t remember the name and I’m away from my computer).

Thanks for the great library!

Dan

dgomes, 2019-04-11:
I accept pull request 😉

danricho, 2019-04-11:
Hahaha I am happy to share my code snippets, but I’m afraid my C++ wouldn’t be up to spec to help write it!

I’ll chuck up the relevant stuff here tomorrow.