dgomes/homeGW #14: ESP8266 "ISR not in RAM"

mcpicoli, 2019-05-27:
Hi,

I successfully compiled and used the code (with the Digoo plugin) and an Arduino Nano board.

Then I tried to move the code to a ESP8266 board ("amica", from a generic supplier), and after adjusting receive/transmit pins in the code, it compiles, but the board is eternally stuck in a reboot loop. The serial monitor shows, repeatedly:
```

>>>stack>>>

ctx: cont
sp: 3ffffda0 end: 3fffffc0 offset: 01b0
3fffff50:  0001c200 0000001c 00000000 00000003  
3fffff60:  feefeffe 00000001 3ffee4f0 3ffee548  
3fffff70:  3fffdad0 3ffee4b8 00000004 40100452  
3fffff80:  0001c200 3ffee4b8 00000004 40201d29  
3fffff90:  feefeffe 00000000 3ffee4f0 402010af  
3fffffa0:  feefeffe 00000000 3ffee518 40202740  
3fffffb0:  feefeffe feefeffe 3ffe853c 4010073d  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v8b899c12
~ld
**ISR not in IRAM!**

Abort called
```

I believe the key here is the "ISR not in IRAM" part. In other projects, when I ran into this kind of problem, it was solved by adding the "ICACHE_RAM_ATTR" flag to the ISR routine, but this routine is part of one of the libraries involved and not in the "user" code.

Therefore, if possible, I would like to ask the devs to add this flag to the optimal routine, given that RAM is scarce in the ESP8266.

Arduino IDE is version 1.8.9 (Windows 10) and the library version is the latest as of 2018-09-15.

Thanks,

dgomes, 2019-05-27:
PR welcomed :)

rplc, 2019-06-23:
Currently have the same issue after updating the esp board library.
My stacktrace (just if needed):
`0x4010044a: __attachInterrupt(uint8_t, voidFuncPtr, int) at /home/rplc/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/core_esp8266_wiring_digital.cpp line 206
0x40201401: HomeGW::setup(unsigned char) at /home/rplc/Arduino/libraries/homeGW-master/homeGW.cpp line 78
0x402010a8: setup() at /home/rplc/Desktop/projects/binaryclock/adruino/DigooReceiver/DigooReceiver.ino line 18
0x402023d4: loop_wrapper() at /home/rplc/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/core_esp8266_main.cpp line 122`

rplc, 2019-08-09:
FYI: With the merge of #15 this issue is solved and can be closed.

dgomes, 2019-08-09:
thanks for the heads up! :)

closed 2019-08-09 by dgomes

rplc, 2019-08-09:
Thanks for the quick merge