dgomes/homeGW #16: Less false positives for Digoo hardware

rplc, 2019-08-09:
Hi,
I've registered a digoo receiver and I get a lot of false readings in my area here. In the following code sample I've added a few sanity checks that seem to exclude all false positives.

```
uint8_t digoo::isValidPacket(uint64_t ppacket) {
  uint8_t id = getId(ppacket);
  uint8_t firstCheck = (ppacket >> 24) & 0x4;
  uint8_t secondCheck = (ppacket >> 8) & 0x0F;

  //check for valid id and control bits, see protocol specs https://github.com/aquaticus/nexus433#nexus-protocol
  if (id != 0 && firstCheck == 0 && secondCheck == 15) {
    return OK;
  }
  return INVALID_SYNC;
}
```

But I am by no means a C++ programmer, so please improve the code ;)

Tested with a DG-R8H and DG-TH8805 transmitter.

Greetings Alex

dgomes, 2019-08-11:
I think the "secondCheck" is a good idea, but I don't think the id and "firstCheck" can be applied to everyone...

rplc, 2019-08-11:
You might be right. Did some digging, "firstCheck" seems to indicate if the transmission was forced (via a button press or so) (see [here](https://forum.pilight.org/showthread.php?tid=3351&pid=22798#pid22798)).