dgomes/homeGW #21: Constant interrupts

swiergot, 2020-06-04:
Hi,

I'm trying to get this library working on my D1 mini (esp8266). I've tried with RXB6 and MX-RM-5V. While I don't have Digoo DG-R8H yet, I was hoping to make some progress with an EMOS temperature sensor.

But the issue I'm having is that interrupts are being generated constantly, regardless of whether it's RXB6 or MX-RM-5V. There doesn't seem to be any packet validation, so the serial console keeps displaying nonsense:

[image]

If I get it right, interrupts should be generated only when the pin changes from 0 to 1 or from 1 to 0. The fact that interrupts are generated constantly suggests that something is transmitting all the time which I think is simply impossible.

Any ideas what my problem is?

Thank you.

dgomes, 2020-06-04:
Hi

If the radio you used is very sensitive it will pickup all sorts of noise from the environment and this can surely happen.

swiergot, 2020-06-04:
Thank you for reply. What can I do in such case? Try to search for some patterns? What radio do you use?

dgomes, 2020-06-04:
Play around with basic filters: https://www.arrow.com/en/research-and-events/articles/using-capacitors-to-filter-electrical-noise

I use a cheap radio bought on aliexpress, some times they are good some time they are rubbish 🤷‍♂️