dgomes/homeGW #23: ISR_PREFIX

danilkorotkov, 2021-03-12:
HomeGW.cpp correction to
#if defined(ESP8266)
    #define ISR_PREFIX ICACHE_RAM_ATTR
#elif defined(ESP32)
    #define ISR_PREFIX IRAM_ATTR
#else
    #define ISR_PREFIX
#endif

dgomes, 2021-03-12:
can you open a PR :) ?

danilkorotkov, 2021-03-12:
im a lazy nube )

danilkorotkov, 2021-03-13:
https://github.com/danilkorotkov/Digoo-Homekit
Used your lib. Many thanks)

dgomes, 2021-03-13:
Good work :)

Still inviting you to open a Pull Request as I have no clue what needs to be corrected

danilkorotkov, 2021-03-13:
I’m using GitHub as a file storage only, sorry)
HomeGW.cpp
Lines from 23 to 27 should be
#if defined(ESP8266)
#define ISR_PREFIX ICACHE_RAM_ATTR
#elif defined(ESP32)
#define ISR_PREFIX IRAM_ATTR
#else
#define ISR_PREFIX
#endif

closed 2021-03-13 by dgomes