dgomes/homeGW #30: Understanding packet detection to improve reception

TimelessNL, 2024-02-14:
Hi,

First I must say great project, with this library I was able to integrate some Prologue sensors with ESPHome in HomeAssistant.
Yet I do have some questions if I may.

According to [the documentation in rtl_433](https://github.com/merbanan/rtl_433/blob/f88bf685787638c54c87411fd60888037d165c43/src/devices/prologue.c#L21) there is a mostly static `type` parameter in the sensors transmission. Checking for this nibble greatly reduced the amount of false positives. This was accomplished by only caching _correct_ packets.
```
void weather::processPacket() {
        uint64_t receivedPacket = 0;
        for(unsigned i=1; i< bitsRead; i++) {
                unsigned duration = this->timings[i];

                if(duration > weather::ONE) {
                        receivedPacket = receivedPacket << 1;
                        bitSet(receivedPacket, 0);
                } else if(duration > weather::ZERO) {
                        receivedPacket = receivedPacket << 1;
                        bitClear(receivedPacket, 0);
                }
        }
        // Check for type nibble
        if(((receivedPacket >> 32) & 0b0101) == 0b0101 ||
           ((receivedPacket >> 32) & 0b1001) == 0b1001 ||
           ((receivedPacket >> 32) & 0b0110) == 0b0110  )
        {
                packet = receivedPacket;
        }
#ifdef DEBUG
        Serial.print("~0x");
        Serial.println((long) packet, HEX);
#endif
}
```

Unfortunately the range, and reception qualities are rather poor using the `SYN115`:
[image]
So I went looking for things to improve but I'm having trouble understanding the code.
I have a couple of questions:
1.  I don't understand how the `packet_size` of [74](https://github.com/dgomes/homeGW/blob/e6a4d511efece8364dcd380c1b4686b8e8f41d83/weather.cpp#L24) is calculated. Looking at the information in [prologue.c](https://github.com/merbanan/rtl_433/blob/master/src/devices/prologue.c#L12) the sensor sends 36 bits 7 times. Would you be so kind to explain how the `packet_size` came to be?
2. Looking at the `detectPacket` function the method [seems to capture packet_size +/- 10%](https://github.com/dgomes/homeGW/blob/e6a4d511efece8364dcd380c1b4686b8e8f41d83/plugin.cpp#L34) before it tries to parse the packet, am I correct to assume that this buffer may contain multiple transmissions as the sensors sends each reading 7 times?
    1. If this is indeed how it works, then I suspect this might be a reason why the reception is lacking? Because when the first transmission is incomplete the entire packet buffer will be offset and therefor none of the bit shifting operations work as intended?

TimelessNL, 2024-02-16:
Ah I didn't even notice the code was that old 😄.
Thanks for helping me understanding the code though, even after all those years.

> so most of the code is about capturing a "good" packet out of the rest

Am I correct to assume it captures a certain amount of data and if the first bit happens to be the start of a packet the `get()` functions and its bit shifting operations will work?

If that is indeed true it might be feasible to scan the bitbuffer for the patterns `0b0101`, `0b1001` or `0b0110` to detect the start of a packet so it can be parsed correctly. I'll give that a try to see if the reception improves.

TimelessNL, 2024-02-17:
**Update:**
Accidentally checked the wrong location of the `0b0101`, `0b1001` or `0b0110` sequence, `32` instead of `28` 🤦
I was also able to use the [remote_receiver](https://esphome.io/components/remote_receiver.html) from ESPHome together with snippets of your code to make it work. This made my situation somewhat simpler and the reception quite decent.

In case if you're interested, this is my config:
```
substitutions:
  sensor_id: '100'
  sensor_channel: '1'

remote_receiver:
  pin:
    number: GPIO23
    inverted: True
  tolerance: 25%
  filter: 250us
  idle: 8ms
  buffer_size: 1kb
  on_raw:
    lambda: |-
      if(x.size() == 74)
      {
        uint64_t packet = 0;
        for(unsigned i=0; i < x.size(); i++) {
          if(x[i] > 3000){
            packet = packet << 1;
            bitSet(packet, 0);
          }
          else if (x[i] > 1000)
          {
            packet = packet << 1;
            bitClear(packet, 0);
          }
        }

        // Verify packet by checking type sequence
        if((packet >> 28 & 0b0101) == 0b0101 ||
           (packet >> 28 & 0b1001) == 0b1001 ||
           (packet >> 28 & 0b0110) == 0b0110  )
        {
          uint8_t id          =      (packet       >> 24) & 0xFF;
          bool    battery     =      ((packet      >> 20) & 0x8)   ? 0 : 1;
          bool    button      =      ((packet      >> 20) & 0x4)   ? 1 : 0;
          uint8_t channel     =      ((packet      >> 20) & 0x3)   + 1;
          float   temperature = float((packet      >> 8)  & 0xFFF) / 10;
          uint8_t humidity    =        packet             & 0xFF;

          ESP_LOGI("433MHz", "id: %d, channel %d, battery %d, button %d,  temperature %.1f, humidity %d", id, channel, battery, button, temperature, humidity);

          if(id == ${sensor_id} && channel == ${sensor_channel}){
            ESP_LOGI("433MHz", "Updating sensor");
            id(sensor_battery).publish_state(battery);
            id(sensor_button).publish_state(button);
            id(sensor_temperature).publish_state(temperature);
            id(sensor_humidity).publish_state(humidity);
          }
        }
      }

sensor:
  - platform: template
    id: sensor_temperature
    name: "Sensor Temperature"
    unit_of_measurement: '°C'
    accuracy_decimals: 1
    icon: mdi:thermometer
  - platform: template
    id: sensor_humidity
    name: "Sensor Humidity"
    unit_of_measurement: '%'
    accuracy_decimals: 0
    icon: mdi:thermometer

binary_sensor:
  - platform: template
    id: sensor_battery
    name: "Sensor Battery"
    device_class: battery
  - platform: template
    id: sensor_button
    name: "Sensor Button"
```

dgomes, 2024-02-17:
Great work @TimelessNL :) 

Closing this now

closed 2024-02-17 by dgomes

mostafaalsham, 2024-02-17:
Awesome, thank you so much!